hivemapper-foundation/hdc_firmware #155: Investigate Wrong Resolution Uploads

punov, 2023-08-16:


punov, 2023-08-18:
HDC-S cameras keep creating 1920*1080 imagery even on updated firmware.

punov, 2023-08-25:
Fixed here: https://gitlab.com/hivemapper/hivemapper-hdc-os/-/commit/6e7d31fc37ef10c16d0a9cb55376423600fb0af9

punov, 2023-09-19:
Jurcek is going to provide the fix tomorrow,
@ste00martin I will give you more details on how to get this thing out for testing

punov, 2023-09-21:
@ste00martin do you know if we have any updates on this topic?

ste00martin, 2023-09-21:
Yes https://hivemapper.slack.com/archives/C05HPP79MPC/p1695070105998399

ste00martin, 2023-09-21:
Jurcek has also mentioned creating a new image here to fix this: https://drive.google.com/file/d/1poMVWi6GAKSjE2xNjZyiovDaclLqKX-1/view?usp=drive_link

I don't have permissions to see this drive folder

punov, 2023-09-21:
@ste00martin What you can do is:
- request an access to the file from Jurcek
- download image
- ssh to camera, copy an image to it
- install it using `mender -install <image>`
- reboot the camera, connect to it again
- systemctl stop odc-api, that will stop ODC API from preventing camera from running
- systemctl start camera-bridge, that will initiate the frames being created in /data/recording/pic folder
- check that resolution is 2024 * 1024
- systemctl restart camera-bridge, restart the bridge couple of times
- and validate that all new frames coming with the same resolution. 

Can you please check this?

You can also do `mender -commit` command, if you want this firmware to stay on your camera. Otherwise it will be gone after second reboot

ste00martin, 2023-09-21:
it works - I'm testing to verify that it works for usb stick later

jayalberts, 2023-09-22:
still need to verify that usb still saves to 4k video

jayalberts, 2023-09-25:
Stefan validated it works for saving 2k images. Haven't validated 4k video works yet.

Testers reporting some videos are 0kb.

jayalberts, 2023-09-27:
New build for 3.5 should have config changes. To be validated.

closed 2023-10-05 by jayalberts