hivemapper-foundation/hdc_firmware #167: Add USB stick telemetry

jayalberts, 2023-09-21:


evanmoss, 2023-09-26:
Simple solution ideas: just have a cron job that does:

Both require the odc-api to be aware if a flashdrive is plugged in, which it should know how to do 

Via ODC-API **if flashdrive is plugged in**:
1. list directory of images (jpg files that aren't folders)
2. group by days
3. mv files to appropriate folder

Simple solution idea for (2): 
Via ODC-API **if flashdrive is plugged in**:
1. Every time the odc-api scans a imu or gnss file, just copy it if it doesn't exist on the flashdrive

evanmoss, 2023-09-26:
cc @punov for situational awareness on aboe

punov, 2023-09-27:
@evanmoss sure, makes sense to me!
We'll start with #2, I will think a bit on #1: 
even listing USB directory takes a lot of time, so I have couple ideas for this one how to make it in optimal way
.

jayalberts, 2023-09-29:
still investigating.

jayalberts, 2023-10-02:
focus on moving images first, then back to location data

HariHive, 2023-10-03:
for images: PR: https://github.com/Hivemapper/odc-api/pull/95

jayalberts, 2023-10-03:
usb is crashing odc-api and hari is implementing and testing a fix.

jayalberts, 2023-10-05:
hari working through PR comments

ODCAPI is trying to move images again that have already been transferred, still need to investigate. Doesn't happen on small amount of images, only larger amounts.

jayalberts, 2023-10-10:
@HariHive to add validation steps

closed 2023-11-10 by jayalberts