hivemapper-foundation/hdc_firmware #193: configurable USB writing FPS

jayalberts, 2023-10-17:


evanmoss, 2023-10-17:
@HariHive Here is a simple approach to reduce write throughput to USB by 50%
```
bool skippedLastFrameForUSB = false;

if (!skippedLastFrameForUSB) {
  skippedLastFrameForUSB = true;
} else {
  // do write to usb
}
```

jayalberts, 2023-10-26:
got build file yesterday from Alexei, started testing.

try to do USB 5 FPS and hivemapping 10 FPS? See if it will work.

jayalberts, 2023-10-27:
dropping every other image didn't help with FPS (~2). Tried every 5th and FPS was closer to 10. 

Writing speed of USB matter. Validate with JT's USB.

Next step: more testing and validate with metrics.

jayalberts, 2023-12-08:
masaya has test written for this

jayalberts, 2024-01-04:
archive this as duplicate of https://github.com/Hivemapper/camera_bridge/issues/9

closed 2024-01-04 by jayalberts