hivemapper-foundation/hdc_firmware #208: HDC stops recording randomly

jayalberts, 2024-01-23:


jayalberts, 2024-01-23:
Mazar investigating logs, might not be related to changes he made.

@jayalberts double check if USB plugged in and exact timestamps.

jayalberts, 2024-01-23:
All times PST

Eddie stopped recording 1/19 approx 11:20-11:40 and 1/18 approx 10:40-11:00.

Koen stopped recording on 1/20 sometime before 2:30 PST

Gator stopped recording on 1/20 sometime probably around or after 9:00 PST

Both Gator and Koen reported app crashing when trying to send DB. Seems to point to some issue with camera DB perhaps?

mazar-farran, 2024-01-23:
Did Koen and Gator use a USB?

mazar-farran, 2024-01-23:
Also, to be 100% sure, all three testers reported the camera view was still showing images?

Masaya-RT, 2024-01-23:
Today I drove on 3.9.1 for about an hour, with USB attached.
- USB recorded normally
- FrameKms uploaded normally

Caveat - Didn't clock several hours like the community testers

jayalberts, 2024-01-24:
@mazar-farran 

Koen: confirmed 3 lights on dashcam and image preview on app
Gator: confirmed 3 lights on dashcam and image preview on app
Eddie: confirmed 3 lights on dashcam but he doesn't use dashcam tab (only map) so can't determine if preview was showing

mazar-farran, 2024-01-25:
@punov image preview on app indicates that camera bridge is unlikely to be the problem.  I found some gaps in the DashcamPackedFrameKm updates on Mode (attached in slack).  But those times also do not seem to all line up with the reported recording times.  Could you take a look at this? Thank you

great-periwinkle-ice gap between 12:32 and 14:04 UTC
hopeful-brown-bluebird gap between 19:20 and 19:40 UTC (as reported) (logs show "TRAVELLED TOO FAR" and "Not enough frames to add yet" messages).
sharp-champagne-pig gap between 20:11 and 22:42 UTC, another smaller one between 17:02 and 17:30 UTC

jayalberts, 2024-01-26:
Another user (Dragon666 on Discord) to investigate with same issue. Collecting more info now.

jayalberts, 2024-01-29:
Mazar investigated further, no smoking gun.

jayalberts, 2024-01-30:
No common cause, all slightly different. At least one user with reliability of GNSS logger.

jayalberts, 2024-03-05:
This might have been related to db issues which were resolved in 4.2. Closing out

closed 2024-03-05 by jayalberts