hivemapper-foundation/hdc_firmware #215: HDC ML Edge Detection

Masaya-RT, 2024-02-15:
Deliverables:
- HDC Firmware with ML Edge Detection model (People & Vehicles)

Requirements:
- Killswitch using `isDashcamMLEnabled` FeatureFlag
- No performance loss on HDC when ML is active
- ML performance is maintained during drives regardless of speed or duration

Masaya-RT, 2024-02-15:
Expect first internal build around noon 2/15/24

jayalberts, 2024-02-16:
Alexei testing today, pushed a fix already. 

Weirdly promising so far. Seeing dynamic switching of grids.

USB writing slightly compromised? 2-3 FPS

Need to do visual checks. One more day of work, internal testing. CTP next week.

Masaya-RT, 2024-02-21:
QA 4.2.1 drive:
- E2E experience is fine
- Gaps in drive, but not too many
- Some small FrameKms but not too many

Deployed 4.2.1 to CTP and ran Audit
- Vehicles are blurred or not blurred, some 50/50 in the same image
- Some images have both blurred and non-blurred vehicles together

closed 2024-03-04 by jayalberts