hivemapper-foundation/hdc_firmware #216: Investigation of model that suits

jayalberts, 2024-02-22:


jayalberts, 2024-02-22:
Expand current audience to all CTP

jayalberts, 2024-02-23:
misses still there but model is performing better.

jayalberts, 2024-02-26:
Nick is working on grid training, then another round of testing

Masaya-RT, 2024-02-27:
Determined that 3x3 grids are no-go. All collections up to 4.2.5 have critical misses.

Good news: 2x2 Max solution seems to be viable. Only requirement is to switch to 12m (instead of 8m) at ~50mph+ minimum
- i.e. 8m @ <50mph, 12m @ > 50mph

Will figure something out for camera-bridge pending Evan's signoff

Determined 50mph: 60mph = 1mi/min -> Every frame should be processed faster than 0.3s. 
- If we want everything in <0.3s, we need lower resolution.
- 12m switch yesterday @ 55mph is hard to sustain for 30+ seconds

Masaya-RT, 2024-02-27:
Addressing FPS drops in camera today. Filtered the 4.2.5 check for Masaya to validate