hivemapper-foundation/hdc_firmware #217: sensor data batching and concatenation problem

jayalberts, 2024-02-23:


jayalberts, 2024-02-26:
problem is solved but uncovered bigger problem:
GNSS message waterfall getting assigned same system time
revisiting how we do system time

Masaya-RT, 2024-02-29:
Fighting with FPS recently, one more deep-dive rabbithole:
- Learned a lot, found we do a lot of redundant steps in camera-bridge
- Most important thing to happen is we need a constant 10fps @ 2k res frames ALWAYS
- Everything else we do should not affect this baseline -> results in bad frameKm chunking -> bad everything else
- When USB connected, does not write to USB. Writes 2k + 4k but never saves -> got rid of this
- A lot of historical logic, 10x/sec to logs
Overall, getting rid of a lot of stuff that does not help 10fps@2k
Only speed change (55mph above and below) is what affects the fps
- Most of today's drive is now 83s and 125s (with small chance of chunks during speed transistions)
- Will push to testing to support parallel USB testing

TL;DR a lot of Camera-Bridge logic takes away from our 10fps@2k flow
- Alexei will see where he can reimplement the other logic, possibly as another service

closed 2024-03-04 by jayalberts