hivemapper-foundation/hdc_firmware #223: Testing new models

jayalberts, 2024-03-04:


jayalberts, 2024-03-05:
seeing huge boost in performance. will help at lower speeds.

Masaya-RT, 2024-03-07:
Did a QA drive using HDC FW 4.4.10.
- Also deployed to 2 city-heavy CTP drivers
- Pending visual validation

jayalberts, 2024-03-07:
recent metrics are optimistic for 800 model
although a little slower
removed CPU limiters to see what performance we can squeeze, looks stable and metrics ok
collects debt on highway but need more data
working on visual validation today

Masaya-RT, 2024-03-08:
Pushing camera to the new limit
New big models are keeping up with the pace "just fine" (not good)
- everything else is deteriorating as a result (GNSS and FPS reduction)
- Ended up being more chunking on FrameKm size (ML OK, chunking bad)
- Deteriorations seen across 10 Users (CTP)
Evan: Can we further limit the CPU usage? Yes, but will probably be critically slow
- Cap inferrence to CPU and leverage other resources for other tasks?
- Make these configurable for rapid experimentation

n800 is still being validated - needs more scenarios

Masaya-RT, 2024-03-08:
Backlogs are OK, prioritize ML + FPS

closed 2024-03-21 by jayalberts