hivemapper-foundation/hdc_firmware #243: Improve kalman filter for gnss

mazar-farran, 2024-04-16:


Masaya-RT, 2024-04-16:
Basic sanity check on GNSS raw data before adding to Kalman filter -> Alexei drove but still seeing issue

-> Investigating; Checking what parameters to include in data
- Alexei found reason for problem this morning: Testing updated logic overnight, found prob still exists.
-> spent more time this morning. Long story short, we have mix of data incoming. DoP is coming separately from location, and data logger mixes both. DoP and Location mix results in invalid data -> even with changes, these messages are still making it into the DB -> resulted in the same issue we see now

mazar-farran, 2024-04-18:
Alexei found the root cause of the bad results.  The kalman filter was being initialized to 0 instead of using the latest data, so whenever the datalogger was reset, there would be a period where all gnss data was off until it had enough data to offset the initial 0 value.

closed 2024-04-29 by mazar-farran